roarfred/AmsToMqttBridge #7: Alternative electrical circuit made with KiCad and using TSS721

hlovdal, 2018-01-21:
Hi. I have started on designing an alternative circuit. The main planned differences are:

* The schematic and drawings are made with [KiCad](http://www.kicad-pcb.org/) (free GPL tool), which brings the benefit that everyone can change and update the schematics and PCB<sup>1</sup>.
* Focus on robustness and reliability. I do not trust the simple voltage divider bus signal converter and want to use the industry standard [TSS721](http://www.ti.com/lit/ds/symlink/tss721a.pdf) chip, fully optocoupled to avoid any ground loops.
* Available as an Arduino shield to be a flexible building block in whatever monitoring system you want to make.
* Support for sending data over wire (serial/usb/ethernet).
* Support daisy-chaining.

The question is how to cooperate on this. I can create a separate repository that you just link to or use as a submodule. Or I can fork this and make a pull request into a Electrical2 directory or better, reorganise the Electrical into containing two subdirectories.

What do you prefer?

---------

<sup>1</sup> Currently you have a PCB that cannot (easily) be modified and you have to post notes like "R8 should not be used" instead of just updating it. That is not very good.

xibriz, 2018-01-21:
I dont have it here on my phone, but one resistor needs a higher value and one need to be added for this to work on Kamstrup.

xibriz, 2018-01-21:
[image]

roarfred, 2018-01-25:
@hlovdal, great initiative! I totally agree on your comment on the rather useless PCB design. If you make a pull request, I'd love to include something better in this project!

hlovdal, 2018-01-31:
[Fork](https://github.com/hlovdal/AmsToMqttBridge) made, and pull requests created. Closing issue.

closed 2018-01-31 by hlovdal