roarfred/AmsToMqttBridge #10: DLMS/COSEM

Ben1902, 2018-02-01:
Hei!

Vil anbefale OpenMUC til logging/lagring av verdier.

www.openmuc.com

Open source logger som har DLMS/COSEM driver, og utvikles fortløpende. Kan kjøre på både windows og linux (f.eks raspberry pi). Har også en fin REST-service for data til eksterne systemer og logging, samt kan konvertere data med gateway extension til andre protokoller som BACnet/IP, modbus, MQTT osv.

Mvh
Benjamin Kruger

roarfred, 2018-02-01:
Kan du utdype? Prosjektet er for å rapportere AMS data til MQTT, og skal kunne kjøres på Arduino/ESP...

Ben1902, 2018-02-01:
Ja,
Først må jeg si jeg setter virkelig pris på kartleggingen du har rundt det nye HAN grensesnittet! 

Jeg ble litt vel engasjert da jeg skrev inn denne "issuen" tidligere, og skjønner nå at du har et poeng med ESP og den trådløse biten. Alt til sitt bruk.

Min bakgrunn er mer logging, og jobber med programmering av dette. Min tanke blir fort at dersom det virkelig er DMLS/COSEM som protokoll på HAN, om det da ikke vil være bedre å benytte f.eks denne openmuc driveren i stedet for å finne opp kruttet på nytt? Samme med enkle MBUS til TTL/RS232/RS485 kretser. De finnes allerede, og det er ingen vits i å gjøre samme feil+analyser på bitnivå som alle andre som har programmert i x antall timer, om du skjønner tanken. Skal heller melde meg inn i det automasjonsforumet og bidra der, ser det blir feil fora å skrive om andre produkter her :) Beklager!

/Ben

roarfred, 2018-02-01:
No worries!

Tanken er god, og om en finner et bra DSLM/COSEM bibliotek som kjører på Arduino, så hadde det vært helt supert å få utnyttet det, fremfor egenprodusert kode. Det jeg er mest skeptisk til i dette tilfellet er å gjøre en dependency mot en ekstern tjeneste. Vil gjerne at systemet skal kunne fungere også uten internett, ihvertfall uten tjenester som en selv ikke har fullt eierskap og kontroll på.

Takk for innspill uansettt! Jeg closer denne issue, så kan vi evt. ta ting videre på hjemmeautomasjon.no

closed 2018-02-01 by roarfred