roarfred/AmsToMqttBridge #21: Reprogrammering

KarlNorway, 2018-02-13:
Hei. Sitter å leser gjennom tråden din (https://www.hjemmeautomasjon.no/forums/topic/1982-lesing-av-ams-data-amshan-iot/ )

Og ser du legger inn knapper for å reprogrammere ESPen. Har du prøvd Arduino-OTA? Den funker kjempe bra. Raskere og enklere å reprog, kan gjøres in-situ. Kan også anbefale WiFiManager. 

Jeg kommer til å benytte meg av denne mot Aidon måleren. Kjører elbil så det kan være en ide å styre ladeeffekt basert på en gitt maksgrense og forbruket i nuet.

Bullhill, 2018-02-13:
Ville da kansje vært greit å bruke https://iotappstory.com/ for å spre updates også.

roarfred, 2018-02-27:
Takk for input, @KarlNorway og @Bullhill!

For et mer ferdig produkt er OTA updates fornuftig. Her finnes flere alternativer, og de skaper som regel en dependency mot noe mer komplekst. Formålet med dette prosjektet var i hovedsak å få til et elektrisk design og et arduino bibliotek tilpasset HAN porten, slik at disse komponentene fritt kan brukes innenfor valgfri IoT platform av andre. Er fremdeles ikke helt i mål med dette, så det får nok forbli første prioritet.

closed 2018-02-27 by roarfred