roarfred/AmsToMqttBridge #25: Make page for contributions

roarfred, 2018-03-05:


Geving, 2018-06-20:
I just wanted to drop a tip for those who have issues with the WiFi signal strength inside the breaker cabinet; Get yourself an ESP-07 instead of the ESP-12. They both have the same ESP8266 chip, and the same form factor, but the ESP-07 has an U.FL Connector for an external antenna (as well as a proper ceramic antenna instead of only a PCB track like the ESP-12 has).

An external antenna using a U.FL connector would only need a 3-4mm hole through the cabinet wall, or you might just close the door over the thin antenna cable.

@roarfred Please feel free to leave, remove, replace or rewrite this text as you see fit.

stenjo, 2018-07-10:
Yes, this will work. The bottom pins are not in use on the pcb. I'm using ESP-12S that also does not have the bottom pins. Works perfectly :-)

DIYglenn, 2018-07-11:
Great! The ESP-07 seems to be the best board for this application because of the better antenna, and the possibility of an external antenna.

[image]