roarfred/AmsToMqttBridge #27: OTA updates

roarfred, 2018-03-05:
Listen for specific MQTT message, with url for binary to update from

xibriz, 2018-04-09:
Something like this:

```
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

const int FW_VERSION = 1011;
const char* mqtt_topic_fw_info = "esp/test/info";

void checkForUpdates(String fwImageURL) {
  HTTPClient httpClient;
  
  String newFWVersion = httpClient.getString();
  
  mqtt.publish(mqtt_topic_fw_info, ((String)"Current firmware version").c_str());
  mqtt.publish(mqtt_topic_fw_info, ((String)FW_VERSION).c_str());
  
  mqtt.publish(mqtt_topic_fw_info, ((String)"Preparing to update").c_str());
  
  mqtt.publish(mqtt_topic_fw_info, ((String)"Downloading new firmware.").c_str());
  mqtt.publish(mqtt_topic_fw_info, fwImageURL.c_str());
      
  t_httpUpdate_return ret = ESPhttpUpdate.update( fwImageURL );
  
  switch(ret) {
     case HTTP_UPDATE_FAILED:
       mqtt.publish(mqtt_topic_fw_info, ((String)HTTP_UPDATE_FAILED).c_str());
       mqtt.publish(mqtt_topic_fw_info, ((String)ESPhttpUpdate.getLastError()).c_str());
       mqtt.publish(mqtt_topic_fw_info, ESPhttpUpdate.getLastErrorString().c_str());
       break;
    case HTTP_UPDATE_NO_UPDATES:
       mqtt.publish(mqtt_topic_fw_info, ((String)HTTP_UPDATE_NO_UPDATES).c_str());
       break;
  }
    
  httpClient.end();
}
```

roarfred, 2018-04-10:
We should also include a initial MQTT message, once connected to the MQTT, and include the FW_VERSION. This will prove the update succeeded. It is important to know the ESPhttpUpdate.update call will immediately cause a reset once completed (on success), so there is no option to report this status inside the update code.