roarfred/AmsToMqttBridge #40: Improvements for Arduino code

xibriz, 2018-04-07:
I suggest the following improvements on your "boot as AP first" code.

- Set the MQTT port to 1883 by default
- In addition to the "Boot as AP on GPIO 0, add a "Clear EEPROM" function on another GPIO

And of course, FOTA :)

roarfred, 2018-04-08:
(F)OTA is requested in #27, by using a specific MQTT message with a URL from where the device can download a new firmware. Are there better ways? like directly uploading a file from a browser, or are we then just making a mess having to parse multipart form-encoded content?

roarfred, 2018-04-10:
There is an implementation that is very close to your suggestion in a project I did a while ago here: https://github.com/xnsense/xns-client/blob/master/xnsclient.cpp#L350