roarfred/AmsToMqttBridge #45: power

mo8Zomo0, 2018-04-21:
Hi! 

Sadly there is no power outlet close by the power meter so I would primary look at a 
self contained solution and then for one which has some good power saving so it could
run a longer time on (rechargeable-)batteries.

From how I see it the only power source here is the USB port. Is there a way to power it
via HAN, somewhere I read that this port should also be able to supply some power.

roarfred, 2018-05-03:
Sorry for the late reply. You can find power capabilities in the PDFs in the documentation folder. Unfortunately, the three known meters differ so much on this parameter, so I don't see it possible to feed an ESP8266 from it. However, power shouldn't be too far away, and if you don't want to install a power socket in your fuse box, you might be able to extend the HAN/RJ45 cable to a place where you have easier access to both power (and WiFi)?

alekslt, 2018-05-11:
I managed to convince the utility guy installing the new meter that it would be a good idea to drill a cable hole for the HAN-cable, so I have a small hole for running a network cable through to have the mqtt-bridge outside the fuse cabinet. :) They should all do this by default, since the few nerds that want to actually use the HAN-connection in the future will propably struggle a bit like mo8Zmo0

ec1oud, 2021-12-21:
Is there still no fork for a bus-powered design, even now that Tibber sells one ready-made?  I'm reluctant to try theirs because 1) it has vendor lock-in of course (data is sent first to their server, and then I can only use a cloud API to read it myself; and 2) trying to get a wifi connection from inside a metal box, a bit far from the router, sounds ridiculous to me.  But it's easy to buy cheap flat RJ45 ethernet cables, so maybe I can just close the door on such a cable and mount the interface outside the cabinet.  Alternatively if there was such a thing as an RJ45 IDC connector, I could use actual ribbon cable... even thinner most likely.

I've got an Aidon meter, and their HAN interface document says it provides 30mA at 24V, which sounds like it ought to be enough.  They even documented how to charge up a capacitor in case the peak load is higher.  https://www.nek.no/wp-content/uploads/2019/02/Aidon-HAN-Interface-Description-v11A-ID-34331.pdf

ArnieO, 2021-12-21:
@ec1oud : Is this what you're looking for? [Pow-U](https://amsleser.no/hjem/7-pow-u-universal.html) sold from my webshop www.amsleser.no

ec1oud, 2021-12-21:
Sounds good.

Also I found out that the Tibber Pulse can be reconfigured.  https://github.com/iotux/ElWiz