roarfred/AmsToMqttBridge #53: Sadly, the author has passed away :(

svenove, 2018-08-10:
Sadly, the author of this project has passed away in April. :(

He's done a great job with this project and many people are using it!
It would be great to continue this project in a fork, but with 10 forks already (and probably more will come), it's difficult for people to know which one is "active" and in continued development.

Are there any "collaborators" added to this project, that has access to modifying the ReadMe without a PR? Based on the commits, @hlovdal might? If so, it would be great to make an "official" fork of this project and point to it in the ReadMe. :)

Here's also an active (Norwegian) forum thread about this project, started by @roarfred himself: https://www.hjemmeautomasjon.no/forums/topic/1982-lesing-av-ams-data-amshan-iot/

DIYglenn, 2018-08-15:
[My fork is focusing on the 'HAN_ESP_TSS721' board](https://github.com/DIYglenn/AmsToMqttBridge), which seems to be the most documented and tested board. I'm currently reorganizing it to make it easier for "end-users" to follow the project.

DIYglenn, 2018-12-07:
Correct, it’s still minor changes, but I’ve reorganized the whole project to focus on that one board, made it easier for non-devs to get what they need. 
I think it would be beneficial to keep it in one fork, for the sake of further development. 
I’ll notify you when I’ve pulled the changes from my side. Hopefully I’ll be able to look more into this when it’s closer to Christmas. Not had any time to continue since summer. 

My thoughts:
Step 1: Organize the folders to archive old development and documentation, and focus on one single board (at least for the moment)

Step 2: Document how to solder and flash the board for newbies (it’s in the works) 

Step 3: Clean up the code, the hotspot isn’t always working, and doesn’t really need to be “fancy” looking (first time setup only).

Step 4: Add OTA updates functionality etc for easy administration, and implement support for more meters than just Kaifa (Nuri). 

I’ll hopefully be able to do my part soon, and then we could start merging changes from the community.

xibriz, 2019-01-03:
I have a much simpler board design (only ~10 parts). I can update my repo with the Fritzing files tonight if I remember.

ohjohnsen, 2024-11-18:
@UPC-PaalDriveklepp I just found this project, and was sad to read that the original designer had passed away. Currently there are almost 50 forks. You wrote back in 2018 that you had some patches for the board. Have you committed these anywhere?

ArnieO, 2024-11-18:
> @UPC-PaalDriveklepp I just found this project, and was sad to read that the original designer had passed away. Currently there are almost 50 forks. You wrote back in 2018 that you had some patches for the board. Have you committed these anywhere?

The maintenance of the project was continued by @gskjold when @roarfred passed away. Since then, @gskjold has done massive improvements and the project lives on today as the amsreader firmware: https://github.com/UtilitechAS/amsreader-firmware
used in AMS readers sold by Utilitech AS: https://amsleser.no

ArnieO, 2024-11-19:
> > The maintenance of the project was continued by @gskjold when @roarfred passed away. Since then, @gskjold has done massive improvements and the project lives on today as the amsreader firmware: https://github.com/UtilitechAS/amsreader-firmware used in AMS readers sold by Utilitech AS: https://amsleser.no
> 
> Awesome that the firmware is still being improved! It seems the hardware itself has become closed source though, as amsleser.no only sell the prebuilt device.

It was me that founded amsleser.no, and later formally teamed up with @gskjold to establish Utilitech AS which now owns and runs the project.

Development and maintenance of both hardware and software is a significant task - and our business model is to sell devices at an affordable price. DIYers are free to use the firmware on personal projects, under a "Fair use" license. That license is as close as you can get to "Open source" while still protecting us from other companies that could use it on hardware they sell. 

A Swiss company did exactly that; sold "copy" hardware with our firmware. They even downloaded the STL file of one of our 3D printed enclosures, changed the logo and sold it for use on their own copycat boards. And they claimed they were perfectly entitled to do so.

So: Correct, we need to be a bit cautious to protect the value of all the work laid down. We do not publish the design files for the hardware, and we only sell prebuilt devices.